nordicsemi/Android-nRF-Connect #1: Where is the source code of the application ?

kirantpatil, 2015-05-14:
Hi,

The git repository does not contain any android source related to the project.

Can you please upload the source code.

Thanks.

reidarma, 2015-05-18:
Hi, and thanks for reaching out to us.

Unfortunately, we do not release the source code for Android MCP.
Currently there is no plans to release the source code. We are however interested in understanding why you desire such access. 
Any feature requests you may have for MCP is interesting as well. 

Best regards,
Martin

kirantpatil, 2015-05-18:
Namaste,

Greetings from India !!!

We want to develop some apps using your chip and we will also release our
app as opensource.

We would like to see how it is written.

Please release the source code as it would help the community to build more
apps and in turn benefit your chip sales only.

Thanks,
Kiran.

reidarma, 2015-05-18:
Hi again Kiran, and thanks for coming back to us.

As I understand your points, you should still not expect to see source code for Android MCP being released.
We simply have not designed this to be released as source code. The app is meant to be a helpful tool to enable developers easily debug and test their implementation. 
Our contributions to ease development of applications interacting bluetooth low energy for Android are mainly nRFBeacon and nRF Toolbox.
They are both released with a 3-clause BSD license. 

I hope you find those sufficient. It is great to see that you also are planning to contribute by open sourcing your applications.
We are looking forward to see the products you are creating.

Best regards,
Martin

closed 2015-07-06 by philips77

philips77, 2025-06-07:
Hello @Anonymous941,
Perhaps I would be able to help you without decompiling it? What issue are you facing and why would you be interested in the source code of nRF Connect?