nordicsemi/Android-nRF-Connect #2: Can't connect to a non-Nordic device

Lytentec, 2015-07-10:
Hy,
is this tool limited to connect to Nordic devices? I try to connect to a different BLE device.
I created a very simple test xml:
`<test-suite description="Very important test">`
`	<set name="EXTRA_ADDRESS" value="C4:BE:84:72:8F:86" />`
`	<target id="myowndevice" address="${EXTRA_ADDRESS}" />`
`	<test id="basics" description="Check basic connectivity of ${EXTRA_ADDRESS}" target="myowndevice">`
`		<connect timeout="4000" />`
`		<refresh />`
`		<disconnect />`
`	</test>`
`</test-suite>`

After starting it it hangs after:
`Removing old result file...OK`
`Copying "sample_test_hrm.xml" to /sdcard/Nordic Semiconductor/Test...OK`
`Starting test service...OK`
`Test started...OK`
`Waiting for the result...`
and never returns.
I don't see any entries in nRFLogger. Using the nRFMasterControl GUI I can discover and connect to my device no problem.
[image]
Thanks in advance
Harald
P.S. I guess, this is not the ideal place to aks these questions, is there a better place? A forum?

Lytentec, 2015-07-10:
Hy,
the issue was not a non Nordic device but the path of the xml file on my adnroid device. For whatever reason, all adb access to "/sdcard/Nordic Semiconductor" fail, probably because of the blank in the path. After changing all paths to "/sdcard/test" and creating this folder on my android device it runs smoothly!
Regards
Harald

closed 2015-07-10 by Lytentec

rideyourstyle, 2016-03-18:
I can confirm, the filename "Nordic Semiconductor" frustrate to start the android service no.nordicsemi.android.action.START_TEST.
I saw the error in the Android Device Monitor form Android Studio.