nordicsemi/Android-nRF-Connect #3: automated testing

schoensn, 2016-02-01:
I´ve tried to use the automated testing function. 
I have edited the sample_test_hrm.xml script and followed the instruciton in the documentation to test the device information service. 

the android debug bridge seems to work. I have testet some commonds before I started the script. In the script i first want to test the "connect" operation, but no result file is generated. it seems that the master control panel (Version 4.03) does not try to connect to my application that runs on a nRF52 DK. the master control panel runs on a sony xperia z android version 5.1.1

on the command prompt the script stops with "waiting for the results"

here is my script file
 ` <test-suite description="Device Information Service">
    
    <set name="DEVICE_INFORMATION_SERVICE_UUID" value="0000180A-0000-1000-8000-00805f9b34fb" />
    <set name="MANUFACTURER_CHAR_UUID" value="00002A29-0000-1000-8000-00805f9b34fb" />
    <set name="MODEL_CHAR_UUID" value="00002A24-0000-1000-8000-00805f9b34fb" />
    <set name="HW_CHAR_UUID" value="00002A27-0000-1000-8000-00805f9b34fb" />
    <set name="FW_CHAR_UUID" value="00002A26-0000-1000-8000-00805f9b34fb" />
    
    <target id="devkit_1" name="Dev Kit 1" address="${EXTRA_ADDRESS}" />

    
    <test id="basics" description="Check basic connectivity of ${EXTRA_ADDRESS}" target="devkit_1">

        
        <connect timeout="4000" />

        
        <refresh />

        
        <discover-services target="devkit_1" timeout="10000" />
    </test>

    <run-test ref="basics" description="Basic tests for DIS">

    </run-test> 
</test-suite>`

do you have any idea why the script does not work?

philips77, 2016-02-02:
Hi,

Did you provide the device address in the command line as described in the first comment of the script?
Do you have the nRF Logger app installed on your Android device? If so, just after you started the app you should get a notification saying that the test has started. When you click it, nRF Logger app will be launched and you'll see the progress, except from the report at the end.
If the notification will not pop up, it means that the script failed to start the service. It may be that it's because of the space in the "Nordic Semiconductor" folder name. Try to alter the test.bat file and replace this name with your name instead. 
I'll do some testing tomorrow.

Best Regards,
Aleksander

schoensn, 2016-02-02:
Hi,

I have provided the device address in the command line with this command:

*test -E EXTRA_ADDRESS C5:7F:B0:8E:AA:91 sample.xml*

I have installed the nRF Logger App on my android device.

and your are right the service is not started. I ve tried to run adb shell
am startservice via the comand prompt:

*adb shell am startservice -a
no.nordicsemi.android.action.START_TEST^C5:7F:B0:8E:AA:91** -e
no.nordicsemi.android.test.extra.EXTRA_FILE_PATH "/sdcard/Nordic
Semiconductor/Test/sample.xml"*

and i get this reply:

[image: Inline-Bild 2]

the app installed on

*sdcard/Nordic Semiconductor/Test/sample.xml*

I hope this information helps you to find a solution for my problem.

Best Regards
Stefan

philips77, 2016-02-03:
Hi Stefan,
First of all, your image is not available :)

But I've started your test on my device and it worked. Make sure you have the latest test.bat file from GitHub. It has changed a little bit some time ago (names of action). From your command I see you should have the latest one, so perhaps it's not the case.

Here's my XML (I've wrapped it in <test-suite...> tag, which was cut(?) in your post):

```
<test-suite description="Stefan's test">
	<!-- 
	    Define devices to test. A Target MAY have a 'name' attribute with a user friendly name and MUST have the 'address' definition with the Bluetooth Smart device address.
	    You MAY specify more than one target but at least one MUST be specified. 
	-->
	<target id="devkit_1" name="Dev Kit 1" address="${EXTRA_ADDRESS}" />

	<!-- 
	    Define tests. Test MUST contain at least one operation elements. Each operation will be performed in order from top to bottom and a result will be logged in the output file.
	    Each test and operation MAY have a 'description' attribute and a target reference. Target may be specified for the test (as a default one for every included operation) and
	    for each operation inside, so you may connect to more than one device in a single test.
	    All operations in the test, except Assert-Services and "DFU" may have a timeout specified using 'timeout' attribute (in milliseconds). The default timeout is 5000 milliseconds.
	-->
	<test id="basics" description="Check basic connectivity of ${EXTRA_ADDRESS}" target="devkit_1">

	    <!-- 
	        First let's connect to the device. The 'description' attribute has been omitted. The tag name will be used as the description. 
	        By setting the 'timeout' attribute to 4000 ms we ensure that the connection attempt will take no longer than that.
	        The 'connect' operation is followed by 1 second break to give Android some time before the next operation.
	    -->
	    <connect timeout="4000" />

	    <!-- 
	        Before we discover services let's make sure that the cache is clear. The services might have changed on the device. The 'refresh' operation must be called after the 
	        device has been connected at least once (a BluetoothGatt object has been created using connectGatt(...) method).
	        Actually there is not need to refresh services in this sample as the 'dfu' operation in the DFU Test (check the next test definition), which is being run prior to this, 
	        does it automatically when completed.
	        The 'refresh' operation is followed by 1 second break to give Android some time before the next operation as it will cause the Android-internal service discovery.
	    -->
	    <refresh />

	    <!-- 
	        In order to read or write something we have to discover available services. 
	        The 'target' is set, but as it is the same as for the test, it could have been omitted.
	        The 'description' has not been specified and therefore it's set to a default value - "Discover-Services".
	    -->
	    <discover-services target="devkit_1" timeout="10000" />
	</test>

	<run-test ref="basics" description="Basic tests for DIS">

	</run-test>
</test-suite>
```

And my command:

**test.bat -E EXTRA_ADDRESS E3:EC:B1:48:6D:C2 issue.xml**

It starts the service using the following command:

**adb shell am startservice -a no.nordicsemi.android.action.START_TEST 
 -E EXTRA_ADDRESS E3:EC:B1:48:6D:C2 -e no.nordicsemi.android.test.extra.EXTRA_FILE_PATH "/sdcard/Nordic Semiconductor/Test/issue.xml"**

I got a notification saying that test is in progress, ans when I clicked it I got this log in nRF Logger:

```
nRF Master Control Panel Test, 2016-02-03
Stefan's test (issue.xml)
A 14:19:50.127 Starting 'Check basic connectivity of E3:EC:B1:48:6D:C2' for 'Basic tests for DIS'
A 14:19:50.171 Connect...
V 14:19:50.196 Connecting to E3:EC:B1:48:6D:C2...
D 14:19:50.204 gatt = device.connectGatt(autoConnect = false)
D 14:19:50.827 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 14:19:50.840 Connected to E3:EC:B1:48:6D:C2
D 14:19:50.855 wait(1000 ms)
A 14:19:51.872 Refresh...
V 14:19:51.881 Refreshing device cache...
D 14:19:51.894 gatt.refresh()
I 14:19:51.907 Cache refreshed
D 14:19:51.920 wait(1000 ms)
A 14:19:52.936 Discover-Services...
V 14:19:52.950 Discovering services...
D 14:19:52.964 gatt.discoverServices()
D 14:19:52.978 [Callback] Services discovered with status: 0
I 14:19:52.987 Services discovered
V 14:19:53.015 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
V 14:19:53.025 Cleaning up...
V 14:19:55.031 Disconnecting...
D 14:19:55.040 gatt.disconnect()
D 14:19:55.053 [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I 14:19:55.061 Disconnected
D 14:19:55.070 gatt.refresh()
D 14:19:55.081 gatt.close()
A 14:19:55.092 'Basic tests for DIS' completed
I 14:19:55.129 Output file saved (issue_result.txt):
Starting 'Check basic connectivity of E3:EC:B1:48:6D:C2' for 'Basic tests for DIS'
   Connect...OK
   Refresh...OK
   Discover-Services...OK
```

The last 4 lines were also copied to the issue_result.txt file which was downloaded to the PC.

schoensn, 2016-02-03:
hi, 

i have tried your xml script.. and downloaded the test.bat again and the service does not start to run.
[image]

when i try to start the service manually i get this error.. 

[image]

Do you have any further idea why the service is not starting in my configuration?

philips77, 2016-02-03:
How about small -e instead of -E?

adb shell am startservice -a no.nordicsemi.android.action.START_TEST  -e EXTRA_ADDRESS "E3:EC:B1:48:6D:C2" -e no.nordicsemi.android.test.extra.EXTRA_FILE_PATH "/sdcard/Nordic Semiconductor/Test/customer_case.xml"
Starting service: Intent { act=no.nordicsemi.android.action.START_TEST (has extras) }

philips77, 2016-02-03:
You can also change 'off' to 'on' in the first line of the test.bat script and remove ' > nul 2>&1' at the end of commands to see the responses from the adb service.

[image]

schoensn, 2016-02-03:
here I try to start the service manually: 
[image]

and here i used the the modified script: 
[image]

in both cases the serivce does not start..

schoensn, 2016-02-03:
you have added the -s option to direct the command to a specific device, but if I have only one android device attached this option should not be necessary.

philips77, 2016-02-03:
Yes, -s is not necessary.

But ss you can see it tries to start the service with act=...START_TEST (which is correct), but also it sets the component name to "Semiconductor/Test/issue.xml", which is not correct. It's all about the space there in the name. Try to change the Nordic Semiconductor everywhere in the script to something without a space.

philips77, 2016-02-03:
I'm not sure if it's the matter of ADB or CMD, but my ADB is in version 1.0.32.

schoensn, 2016-02-03:
i have the same version of the ADB. I will try to remove the blank in the script..

schoensn, 2016-02-03:
without the blank it works. :+1: thank you for your support

philips77, 2016-02-03:
Super! Hope you'll like it!

closed 2016-02-03 by philips77