nordicsemi/Android-nRF-Connect #4: Exception when reading characteristics

evertvr, 2016-03-14:
I get an android exception when I try to read a HID characteristic. It looks like the application has insufficient rights:
`gatt.readCharacteristic(00002a4a-0000-1000-8000-00805f9b34fb)`

`E	16:16:17.745	Exception occurred (Need BLUETOOTH_PRIVILEGED permission: Neither user 10095 nor current process has android.permission.BLUETOOTH_PRIVILEGED.)`

`V	16:16:17.759	Disconnecting...`

philips77, 2016-03-14:
Hi, since Android 5.0 the HID characteristics are secured and only available to the system. This is to protect from password spoofing, as before any app could have enabled notifications on those chars and listen to keys.
At least I think so, as Google didn't explained this.

philips77, 2016-03-14:
This permission is system restricted.

closed 2016-03-14 by philips77