nordicsemi/Android-nRF-Connect #5: WRITE_COMMAND doesn't work

rideyourstyle, 2016-03-22:
Hello
We build a BLE-device and created our own porfile. The Master Control Panel provide a good way to run tests automatically.
I hava a problem with write on a character with the type WRITE_COMMAND. The same write command works with the type WRITE_REQUEST:
```
<write description="Write measure config step 1" type="WRITE_REQUEST" service-uuid="${CONFIGURESCBS_SERVICE_UUID}" characteristic-uuid="${MEASURE_CONFIG_CHARACTERISTICS_UUID}" value="E703E70308E70302000000" />
    <read description="Read the new measure config" service-uuid="${CONFIGURESCBS_SERVICE_UUID}" characteristic-uuid="${MEASURE_CONFIG_CHARACTERISTICS_UUID}">
    <assert-value description="Check the new value" value="E703E70308E70302000000" />
</read>
```
But i can write with WRITE_COMMAND and WRITE_REQUEST when I use the Master Control Panel and write manually (open the profile and write maually)
Are there any known issues?

Thanks and greetings
Peter

philips77, 2016-03-22:
Thanks, I'll have a look at it after Easter.

philips77, 2016-04-06:
Hi, do you still have that problem?
I tested it here and both modes seems to work fine. What if you put ```<sleep timeout="500"/>``` between write and read (I don't know why would it help...).

philips77, 2016-04-14:
Hi again, I did some tests and both writes work fine on my side. I'll need some help from you to replicate this issue.

philips77, 2016-05-03:
Hi, do you still experience this issue?

closed 2016-05-11 by philips77