nordicsemi/Android-nRF-Connect #6: AutoReconnect no longer working

allenbaker, 2016-04-06:
In v3.4.1, the "Connect with autoConnect" option works properly and successfully reconnects to the device if the connection is interrupted.  In at least the last two versions, this functionality is broken and does not automatically reconnect to the current device.  I noticed in logcat that gatt.close is being called, which would explain why it is not working.  I do not see that message in v3.4.1 though.

Just a heads up :)

philips77, 2016-04-06:
Hi, I was told about it already. There will be an update on Friday or Monday with this bug fixed. Thanks for letting us know.

allenbaker, 2016-04-06:
No problem! Thank you for the prompt response.

philips77, 2016-04-14:
Hi, I've just released version 4.1. This problem should be fixed now. Let me know if it's not.
Sorry for the delay.

closed 2016-04-14 by philips77