nordicsemi/Android-nRF-Connect #8: Save characteristic read value to log

alzander, 2016-05-06:
I'm working on an automated test suite, but need to check sensor values that won't always have a fixed value (like a temperature sensor).

Assertions look to be for only fixed values, and not ranges (23-26 or 24 +/- 2), so I can't use those as they are currently.

To allow tests to proceed regardless of value, I'd like to simply save the read value from a characteristic and write the value to a log file so I can later parse that value out and verify it was within a proper range.

So, can a line like below:
<read description="Read TEMP Sensor" service-uuid="${TEMP_SERVICE_UUID}" service-instance-id="0" characteristic-uuid="${TEMP_CHAR_UUID}" characteristic-instance-id="0" />
be used to add:
"Read TEMP Sensor ... 25"
Or something similar? Right now, the assertion will show the read value, but I don't see any way to show it otherwise.

Thanks!

philips77, 2016-05-06:
Hi,
That's not quite possible to do, as the data you receive, send or read may have any format, for example they can be 4 characters, 2 uint16, or one float32. I have no way to decoding them properly.
However, you may specify the expected output of a command:
```
expected="SUCCESS_WARNING_ON_FAIL"
```
and provide the "expected" value in *value* tag. Then, in the report, when the value is different from what you assumed, you should get something like:
WARNING: Value of the characteristic '020304' is not equal to '010203'.

I know it works for text values, but should also work for byte values.
Then you have to parse the output file, search for the data obtained and check them yourself.

I hope that helped.

alzander, 2016-05-06:
I understand the formatting issue. Hadn't tested across a large set of cases, just a unit16 for now.

Your solution is workable though since we'll be parsing things out as it is.

Thanks!

philips77, 2016-05-06:
Cool. Please, try it. If it won't work let me know, otherwise, please close this issue.

alzander, 2016-05-09:
Will be testing this out over this week and will close. I want to try to do some testing with each of those format types above to see how they are actually logged. Can you tell me how the assertion process works with each of those, since there's no method to set the format as it is so unsure how you test during the assertion process.

Another possibility would to add an attribute for formatting so it can be displayed accordingly.

philips77, 2016-05-09:
When you set the value using the *value* attribute (seeing bytes), the app compares all bytes, exact. If they match, depending on the *expected* attr, success, warning or fail are returned.

alzander, 2016-05-16:
Thanks for the explanation. Have initial working version of our testing suite. Will keep testing with different return values, but this works for us.

closed 2016-05-16 by alzander