nordicsemi/Android-nRF-Connect #9: Read MCP version number

alzander, 2016-05-06:
Can you add a feature to be able to read the current MCP version number? Ideally, we'd like to be able to add a Version check along with an assertion at the top of our automated script to ensure either:
a) A specific MCP version is installed and being used on the phone. This allows for distribution of our automated test to reduce problems and conflicts from older or newer releases.
b) A version, or higher, is installed and being used on the phone. This allows for development as adding for great new features from you :)

If the assertion for version or higher isn't feasible, simply writing out the version to a log file would be a great start. This could be a standard line to write out at the top or initiated by a read command. I'd assume this would help you as well if user's submit log files of errors to know what version of MCP is installed.

Thanks!

philips77, 2016-05-06:
Thanks for the feedback! It's a really good idea. For sure I'll do it, although I'm not sure how quick...

philips77, 2016-06-20:
Hi, question. I will log the nRF MCP and phone/talbet version in the first line of the report file and in the log, Is it OK for you?

sourcecoast, 2016-06-21:
That version information is exactly what I'm looking for. Thank you!

sourcecoast, 2016-06-24:
Just saw the version output in the scan issue output. I see:
24-06-2016 16:44:01
Version: Instant Run
Device: Nexus 6P, Android version: N (NPD56N)

Is the Version: field going to be the MCP number and the top row the run time? Assuming Instant Run is just a debug parameter, but want to double check.

Looks great!

philips77, 2016-06-24:
Yes, it will be 4.3.0.

philips77, 2016-07-13:
The nRF Connect 4.3 is now released and should be available on Google Play soon. The docs has been updated here.

closed 2016-07-13 by philips77