nordicsemi/Android-nRF-Connect #10: Log Scan results

alzander, 2016-05-09:
I know the docs mention "The device must be connected to read the RSSI as scanning is not yet supported in automated tests."

Can you give any clues on when scanning may be available?

As mentioned before, I'm working on a functional test suite that uses the automated tool. The idea is that we can automate the scanning of any local units and then test them. In the factory, we can control that only one (or only test) units are nearby. The scan will allow us to read the UUID and then initiate the test on that unit without user intervention.

For my purpose, simply listing the Device Name, UUID and (ideally) the last RSSI in the log is all that would be necessary. The rest I can automate.

The suite is PHP based and I plan to release the test suite on Github. With luck, I'll have a very early/crude stab up next week, if that helps :)

philips77, 2016-05-09:
Hi, you'll have to wait some time for this feature... for sure not in May, and it may be difficult in June... but it's in my backlog and I remember about it.
Could you write what do you expect from such a test? In my mind, user should be able to define a *scan* command and set the assertion (one? more?) with expected adv data. It would be useful if such command could then define a *target* for other commands (connecting to targets that adv with a random address). Is there anything else you'd like to have there?

alzander, 2016-05-11:
What you describe would be a great long-term solution. The assertions and auto-setting of the device based on specific parameters is more than we would need though. My requirement may provide an easier launching point, hopefully.

I'm looking for a command, like:
<perform-scan>
In the log, it would simply display:

> Scan performed... Devices found
> | 1 |  11:22:33:44:55:66 | -45dB | Device name 1
> | 2 |  22:22:33:44:55:66 | -79dB | Device name 2
> | 3 |  33:22:33:44:55:66 | -89dB | Device name 3
> | 4 |  44:22:33:44:55:66 | -100dB | Device name 4

From there, my script will check the log file and can generate the next XML file or command line arguments to call with the UUID found.

If someone isn't using my script, they can still cut/paste from the command prompt to initiate the next test.

Longer term, we'd use the assertion and selection of a device based on criteria syntax, if available.. but for now, it's not something we need and I still think a scan call is useful without it.

Thanks!

alzander, 2016-05-16:
Just wanted to post a link to our project. We are committed to keep releasing updates making your awesome tool even easier to use.
https://github.com/hellonuzzle/bluetooth-fct

There's lots more to do, but the only missing piece is <scan> element to return devices found. Hopefully you can run this up the flagpole at Nordic to get you more resources/time as I think we can make something awesome together.

Thanks!

philips77, 2016-05-24:
This is awesome! Thanks for doing this! :)
I will try to help you with scanning feature but, as I wrote, it may take some time.

alzander, 2016-05-24:
Thank you! It's a great tool for testing and lots of potential with the integration I'm adding. Hope lots of people make good use of it.

Whenever you can get any version of scanning implemented (again, even just logging is all we need), we'll be able to test very thoroughly. Our boards have an LED on them, so our intent is to:
* Run scan
* Offer selection to user to choose device (if multiple fitting the criteria is found)
* After selection, turn on the LEDs using BT and have user confirm proper board
* Run full functional test suite on that board

Giving you an idea of how cool it will be :)

philips77, 2016-05-24:
Very cool indeed ;)

philips77, 2016-06-24:
Hi, so I think I'm done now with the scanning feature. Could you, please, take a look at 
[new scripts.zip](https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/files/332018/new.scripts.zip) and say if you understand everything and if suits your needs? The result file so far looks like this: [sample_test_scanning_result.txt](https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/files/332019/sample_test_scanning_result.txt)

I'm planning to release it in the upcoming release together with renaming the app to nRF Connect. I'll update the documentation on GitHub together with the release.

philips77, 2016-06-24:
Btw, I found out that tests do not work now on Android N preview 4. getInstanceId() method in BluetoothGattService and Characterstic return the Handle number now, not 0+ instance number. This will start working again in the new release. Just FYI.

sourcecoast, 2016-06-24:
The scan results look great to me. Should be able to find and filter out exactly the type of devices we want to connect to and then run tests from there. Can't wait to test

sourcecoast, 2016-07-05:
Just curious if you had an idea of when the new release will be available. Excited to use the new features!

philips77, 2016-07-05:
I hope today. Hurrah!

philips77, 2016-07-05:
Hi, so I was told, that because of the vacation period we have to wait until Friday with the release. We also release the nRF Connect for Desktop and want to make it properly. Sorry :(

philips77, 2016-07-13:
The nRF Connect 4.3 is now released and should be available on Google Play soon. The docs has been updated here.

closed 2016-07-13 by philips77

philips77, 2017-01-20:
Hi, small FYI.
In the next nRF Connect I'll add support for 2 new tags to both automated tests and macros:
```xml
<sleep-if description="Wait for button click" service-uuid="${UI_SERVICE}" characteristic-uuid="${UI_BUTTON_CHAR}" value="00" expected="SUCCESS_WARNING_ON_FAIL" />
```
This wait until the value of given characteristic is equal to the value specified.

```xml
<sleep-until description="Wait for button click" service-uuid="${UI_SERVICE}" characteristic-uuid="${UI_BUTTON_CHAR}" value="01" timeout="10000" />
```
This waits until the value of the characteristic equals the value specified.

Use cases:
- buttons (wait for button to be clicked, released, is not clicked, not released, etc)
- buffer overflow notifications (wait if status-char value = buffer full)
- initialization (waits until char. value changes)

In functionality it's almost like assert-notification (or wait-for-notification in macros) but goes on only when a condition was not met), and also works if the value matched requirements even before, notification does not have to be received.

Any feedback is welcome :)