nordicsemi/Android-nRF-Connect #11: Some question about scan remote devices

GithubRyze, 2016-05-16:
Dear friend:
 i have a  question to ask you.
       i was developer an  android  application to scanning the bluetooth device.And i use android API   mBluetoothAdapter.stopLeScan(mLeScanCallback and  mBluetoothAdapter.startDiscovery(),but i cannot get the bluetooth devices!!!! then i use you nRF-Master-Control-Panel app can found the devices. can you tell how can i get the bluetooth devices!??Doesn't you not use the Google API to develop??  thanks!

mostafaberg, 2016-05-16:
This is not an issue in our app, please post your question on an android developer forum or in our nordic dev zone so we can help you further, or you can just read the code where we do the scanning :)

closed 2016-05-16 by mostafaberg

GithubRyze, 2016-05-16:
How to read  the code where you do the scanning? i can not found the open source code!! please  give me  a linker to your source code !!  thanks !!

philips77, 2016-05-16:
Hi, check the [nRF Toolbox](https://github.com/NordicSemiconductor/Android-nRF-Toolbox) project for info. The methods you use are correct. I also recommend using the [Scanner Compat library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library).

GithubRyze, 2016-06-21:
thanks!! this issue was solved. in some smartPhone need location permission and in android 6.0 should request permission use coding  not  in MainFest.xml