nordicsemi/Android-nRF-Connect #12: MCP should accept/display both ASCII and HEX input/output

frief, 2016-05-24:
It would be nice if MCP would e.g. for Nordic UART Service (or others) be able to send/display arbitrary bytes.

Currently non-printable chars are neither displayed nor can be sent (unless I'm missing something). The app shows "Value:". The log has them though ("value: (0x) 01").

Thanks,

philips77, 2016-05-24:
Hi @frief, you can now do this. When connected, go to menu and deselect "Parse known characteristics". Then you will have the default message builder and the data will be shown as bytes.

frief, 2016-05-24:
good, this helped!)

(Probably would not have found this myself for sending because I'd have assumed that "Parse known characteristics" would apply to receiving only)

philips77, 2016-05-24:
Great. Closing the task.

closed 2016-05-24 by philips77