nordicsemi/Android-nRF-Connect #13: Location of cloned services

mainr, 2016-06-13:
When I select "Clone device's services" menu option, the app informs me of the filename written.

WHERE can I find the file the services are written to? If it is not accessible by a file browser (I'm using Total Commander on Android), how is the stored service file accessed?

I'm assuming the format of this file is human readable...

Thanks.
R. Main.

philips77, 2016-06-13:
Hi, 
The services are saved in local database of nRF MCP. You can load them and then export to a file by selecting Configure GATT Service in the navigation menu (hamburger menu).
Select your configuration there from drop down list and use menu->export option. It will be saved in SD card in Nordic Semiconductor/Server folder.

mainr, 2016-06-13:
Thanks. The process was a bit convoluted, I did manage to extract an XML file of the attributes. I'm dissecting the interface of a device, and having a listing of all attributes is handy.
Regards.

closed 2016-06-13 by mainr