nordicsemi/Android-nRF-Connect #15: Automated Testing - Eddystone and NUS

lauleehong, 2016-06-14:
Hi,

1. How can I check that Eddystone is advertising using the automated testing? I did the following but it doesn't seem to detect:

<set name="EDDYSTONE_SERVICE_UUID"	value="0000FEAA-0000-1000-8000-00805F9B34FB" />
<assert-service description="Check if Eddystone Services exist" uuid="${EDDYSTONE_SERVICE_UUID}" instance-id="0">
		</assert-service>

Also tried the long 128-bit uuid "93FB4E84-CCF1-4ED9-BD24-053267701B25" but doesn't detect as well. I know it is probably the way the nRF Master Control Panel detects Eddystone. Any idea?

2. How should I go about testing out the Nordic UART Service (NUS)? I would like to send a string "ABCD", and ideally my hardware would send a return packet "ACK".

philips77, 2016-06-14:
Hi, so far the automated testing module does not allow for scanning. This feature should be added in coming months. So far you may only connect to a device, discover services and do operations requiring connection.
For the UART service you may do what you wrote. You just need to provide it's Mac address and prepare script that will do what you want. This, as I see, does not rely on scanning or parsing adv packets.

closed 2016-06-14 by lauleehong

philips77, 2017-01-20:
Hi, FYI. nRF Connect supports scanning in automated tests for some time now. You will find more information in the documentation.