nordicsemi/Android-nRF-Connect #16: DFU status when bluetooth disconnect

ShaoyuLee, 2016-06-22:
Hi,

Platform: nRF5_SDK_11.0.0_89a8197
Master Control Panel: V4.1.3

1.If dfu has started and transported some data, then disconnect bluetooth and kill app.
2.The second time to connect start dfu, will return DFU_STATUS_INVALID_STATE.
3.The third time to connect and start will success.

I check the firmware, the firmware has a variable m_dfu_state to mark the state of dfu.

> typedef enum
> {
>     DFU_STATE_INIT_ERROR,                                                           /**< State for: dfu_init(...) error. */
>     DFU_STATE_IDLE,                                                                 /**< State for: idle. */
>     DFU_STATE_PREPARING,                                                            /**< State for: preparing, indicates that the flash is being erased and no data packets can be processed. */
>     DFU_STATE_RDY,                                                                  /**< State for: ready. */
>     DFU_STATE_RX_INIT_PKT,                                                          /**< State for: receiving initialization packet. */
>     DFU_STATE_RX_DATA_PKT,                                                          /**< State for: receiving data packet. */
>     DFU_STATE_VALIDATE,                                                             /**< State for: validate. */
>     DFU_STATE_WAIT_4_ACTIVATE                                                       /**< State for: waiting for dfu_image_activate(). */
> } dfu_state_t;

I 

How to handle this situation?I meet this on our real production with your [Android-DFU-Library](https://github.com/NordicSemiconductor/Android-DFU-Library).
Thanks.

philips77, 2016-06-22:
Hi, 
Could you, please, post your question on the DevZone? It's not really a bug in nRF Master Control Panel or nRF Toolbox.

I don't think something can be done about it. I was told, that adding a delay in the DfuBaseService after connecting to a device solves some Invalid State errors, so it has been added, but maybe it does not solve this issue. Looks like it has to be done 2 times if interrupted.

ShaoyuLee, 2016-06-22:
Thanks.
Maybe I can get a answer from DevZone.

ShaoyuLee, 2016-06-22:
https://devzone.nordicsemi.com/question/84479/dfu-status-when-bluetooth-disconnect/

philips77, 2016-06-24:
If it can't be fixed on DFU Bootloader, I think a solution may be to send a Reset command when Invalid State is received and try again. I'll look into it.

foobar8675, 2016-06-29:
Hi @philips77 , any update?

philips77, 2016-06-29:
There is a DFU update coming late July. Based on the answer you got on DevZone, I will try to implement resuming feature, instead of always starting DFU from scratch. Perhaps it will work and invalid state will not be thrown. Could you try it, BTW? Do a partial DFU, disconnect, and on the second connection, instead of starting DFU again, read number of stored bytes manually and send some bytes as if you were resuming. Check if invalid state will not be thrown then, ok?

If not, I'm also planning to implement retrying in case of 133 error or this one. This should also help it the above one will not.

philips77, 2016-06-29:
@mostafaberg

philips77, 2016-06-29:
Related https://github.com/NordicSemiconductor/Android-DFU-Library/issues/15

foobar8675, 2016-06-29:
@philips77 wouldn't it be safer to always start from scratch and send a reset if invalid state is received? resuming isn't really necessary for my use case since the app / hardware only gets into this state sometimes.

philips77, 2016-06-30:
Hi, the issue should now be "solved" in the DFU Library. I have added a second attempt in case of connection error 133or if Start DFU returns "Invalid State" error. DFU Library version: 0.6.3 on jcenter. 
It will be in the upcoming version of nRF Master Control Panel, or rather nRF Connect. Release date tomorrow or Monday.

foobar8675, 2016-07-01:
Thanks @philips77 . I'm not sure I see what pod version 0.6.3 corresponds to. I can just wait until monday as well.

philips77, 2016-07-01:
https://github.com/NordicSemiconductor/Android-DFU-Library this is there lib used in DFU.

mostafaberg, 2016-07-01:
Just noticed I was tagged, I'll look into that on Monday , however it won't be given high priority, but will do my best to look into a better solution

closed 2016-10-26 by philips77