nordicsemi/Android-nRF-Connect #17: Disconnect fails if app is open

hellonuzzle, 2016-07-07:
Whenever a test ends, a disconnect should automatically happen. If I have the app open on my phone, then the disconnect does not occur.

If I open the Task Manager on the phone and close the app, and then try the same test, a disconnect does occur as expected.

The same behavior happens if I use the <disconnect> tag.

Tested on 4.2 and 4.3

alzander, 2016-07-07:
That was my post above.. from a different account :)

philips77, 2016-07-07:
It's not really a bug... On Android each app, service etc may connect to a device separately. The system keeps a single connection, but each app thinks it's the only one. It gets a disconnect event after calling gatt.disconnect() even if some other app is still connected. The physical connection is open. That is the case here. If you have nRF Master/Connect open (or any other app with GATT server open) with the GATT server enabled and the setting Settings->Connectivity->Show incoming connection set to on, it will connect to your device as a second app. Then the service in automated tests will fail to end the connection and the device will not start to advertise again as Android will still be connected.
To force disconnect you'd have to have a characteristic that triggers disconnect from the peripheral side after writing something. I recommend a graceful disconnect as the default timeout is long, 20 sec.
So the answer is... Try to have nRF app closed during tests, or disable the GATT server there. Or switch this setting off. But still some other apps can choose the same issue.

alzander, 2016-07-07:
Thanks for the explanation. Now that we found out how to make it work, it's not an issue for us, but wanted to point it out if it was some sort of comms problem with the app and your service. Closing this ticket.

closed 2016-07-07 by philips77