nordicsemi/Android-nRF-Connect #19: nRF Connect intercepts connection from another app

michaldrabik, 2016-08-03:
Hello! 

I've stumbled upon an interesting "issue". It occurs when working with (for example) Kontakt.io app and having nRF Connect app opened in the background. When I try to connect to a beacon with Kontakt.io app, I can't because somehow nRF initialises connection by itself.

I've recorded a short video with described behaviour: 
https://drive.google.com/open?id=0B-sOINpYt_YMYUVTckl6aWpTUlk

Why would nRF app initialise a connection by itself when I'm explicitly doing it from another app.

roshanrajaratnam, 2016-08-03:
Hi,

This could be because the GATT Server is enabled on the nRF Connect app. If
you go in to the navigation drawer, select configure GATT Server and
disable GATT Server. As long as the GATT Server is enabled and the nRF
Connect app is running on the background, nRF will take over the
connections.

philips77, 2016-08-03:
Hi, let me add some more info. What Roshan said is true. If the GATT server is enabled in any Android app (including nRF Connect) it will receive a onConnectionStateChange event for ever connected or connecting BLE device. If it then call connectGatt(..) it will connect to it. However, up until N release there was a bug on Android and only one connection callback could have been registered. So when contact.io app initialize connection, the nrf will do it shortly after the device got connected. Konntkt.io received onConnectionStateChange but before it receives onServicesDiscovered the callback will be overwritten so only nRF (or any app with hungry server) will receive competition callback. As i saw in AOSP there is now a list of callbacks implemented, not a single one. But I'd have to verify it.
If you want to have the GATT server enabled you may also disable Show incoming connections in Connectivity settings in nRF to disable calling connectGatt method.

michaldrabik, 2016-08-05:
Thanks for clarification.

closed 2016-08-05 by michaldrabik