nordicsemi/Android-nRF-Connect #21: Taking over connection of other app

vliedel, 2016-10-26:
When the nRF-Connect app is opened, and I connect to a ble device in my own app, the nRF-Connect takes over the connection: you see it being connected in the nRF-Connect app. Is this intended behaviour?

philips77, 2016-10-26:
Yes, it's intended. It lets you debug communication (e.g. incoming notifications and indications) from other apps
However, it only does it when you have the GATT server enabled and the Show incoming connections option set in Connectivity settings.

vliedel, 2016-10-26:
Ok, thanks for the info :)

vliedel, 2016-11-30:
Hi, after disabling "show incoming connections", I still noticed the nrf connect app took over the connection from my app and kept it connected. Is there some other case in which this can happen?

philips77, 2016-11-30:
Than I think you have to disable the GATT server in the app. Go to Configure GATT Server and select the Disable (turn off) icon there.

vliedel, 2016-11-30:
That was disabled already :)

philips77, 2016-11-30:
With GATT server disabled nRF Connect shouldn't take over connections.

closed 2017-04-07 by philips77