nordicsemi/Android-nRF-Connect #22: Make tests runnable from the app UI

eliotstock, 2016-10-26:
The automated testing here looks really powerful and we'd love to use it for desk testing during development as well and end of production line testing. But we don't have a Windows machine in either situation and don't want to add one. It was disappointing to see that, to use this, the phone/tablet has to be connected to a host machine.

Can you not add some UI to the app that allows the user to select an XML file for the test from the filesystem, optionally specify device and extras, then run the test and see results in real time? I think that would make this much more useful.

At the very least, it would be nice to see a bash version of the test script that I can then call from make, or convert to a Makefile.

philips77, 2016-10-26:
Hi, perhaps we may have a Python script in the house. I'll ask tomorrow. This should help, I guess?
As of ui tool there are macros in the app. The XML schema is somehow similar to one in tests, it supports GATT server operations as well, but no scanning nor report at the end.
Also, check #10 and  https://github.com/hellonuzzle/bluetooth-fct

eliotstock, 2016-10-26:
Thanks, a Python script would be a start.

Not sure how the macros help. Are you suggesting I kick off a test with a macro? We would definitely need to scan.

Nuzzle's tool looks interesting, but I'm reluctant to add PHP to the toolchain. Will keep an eye on it.

philips77, 2016-10-26:
You can actually write this script in bash yourself. There are just 2 commands: copying XML to a phone and starting a device + removing the old result and waiting for a new one. You can find a sh script in sdcard/Nordic Semiconductor folder, which is created when the app asks you for a permission on first launch, that starts a DFU operation. You may modify it to start the service specified in bat file. Should be fairly easy...

philips77, 2016-10-26:
Macros also validate data. But if you need scanning or reading result automated tests are what you need.

eliotstock, 2016-10-27:
OK, thanks. I'll do that. I'll probably do it all from Make, without bash or Python.

So, that's really "Make tests runnable from Mac/Linux" we're talking about whereas I still think "Make tests runnable from the app UI" would be a very useful feature. So I'll leave this open, but feel free to close as "won't fix" if you don't agree or can't see it making the roadmap. Thanks.

simonratner, 2017-02-24:
Another vote to allow running tests via mobile app UI.

mbolgiano, 2017-02-25:
Another vote to allow running tests via mobile app UI.

askoropadsky, 2018-02-14:
Another vote to allow running tests via mobile app UI.