nordicsemi/Android-nRF-Connect #23: App fails to ask for local storage permission

eliotstock, 2016-10-27:
I'm using the app for the test script functionality. When I run a test script over adb, the nRF Logger app shows the following error in the logs on the Android device:

```
Parsing test suite failed:
/sdcard/Nordic Semiconductor/Test/ble_out_of_box.xml: open failed: EACCES (Permission denied)
...
```

If I go to Android's Settings, Apps, nRF Connect, Permissions, then sure enough, "Storage" is unchecked. But the app never asked! On Android 6.0 and up, you need to ask at runtime, I think. I'm on Android 6.0.1 on this device.

philips77, 2016-10-27:
It always asks after first launch. And if you don't agree then you have the option available in menu on Scanner tab. It also asks when you want to export a GATT config it a micro to a file. At least it should :)

philips77, 2016-10-27:
It might have not asked you if you updated your Android to 6 and the files were already on SD card...

closed 2016-11-14 by philips77