nordicsemi/Android-nRF-Connect #27: How can we make the connection more stable?

mjohn123, 2016-11-23:
Thank you for giving us the information. I am using your app to connect with my BLE device. It is a really stable connection. Currently, I am following the google sample Bluetooth Le Gatt (BluetoothLeService.java). It provided two option for connection 

 `mBluetoothGatt = device.connectGatt(getApplicationContext(), autoConnect, mGattCallback);`

with `autoConnect=false` for direct connection and `true` for background connection. I am not sure which autoConnect value do you use in your application to make a stable connection. I test both options but it looks same. I am using Galaxy S4, Android 5.1

Thank you

philips77, 2016-11-23:
Hi, I enclosed many Android related questions and answers, including information about autoConnect, here: https://devzone.nordicsemi.com/blogs/1046/what-to-keep-in-mind-when-developing-your-ble-andr/
Hope you find it useful.

mjohn123, 2016-11-23:
Thanks for sharing that document. Actually, I am also using that library. But my application is not stable connection comparison with your app. I am setting `autoConnect=true`. Do you have any suggestion to make the connection between BLE and the phone stable? Thanks

philips77, 2016-11-25:
Well.. I'm actually not doing anything special.
Besides, I print all relevant method calls on the log (including wait(timeInMillis)). 
You could also check the BleManager class from nRF Toolbox ([here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java)). It handles the connection and allows to define an API in a easy way. Can be used in the Service as a single or one of multiple connections. The code there is fairly similar to one in nRF Connect.

closed 2017-01-20 by philips77