nordicsemi/Android-nRF-Connect #28: Enable/Disable Notify is More Important than UI fresh

jasonbu, 2016-12-05:
When I debug my product with nRF-connect 4.8.0

My device notify 100times per second.

so When i enabled the notify. It's almost **imporssible to disable** it. as the UI is always refreshing.

philips77, 2016-12-05:
Yes, that's a know bug/limitation, but I don't know if it can be fixed. I would have to start I ignoring to fast notifications, which would be worse, in my opinion.
What you can do is:
1. Install nRF Logger. This app will store your logs even when nRF Connect will be killed or crashed.
2. Open nRF Connect, connect to your device and enable cccd.
3. The ui will freeze, but then you can click Recents button in Android and remove the app. This will disconnect your device.
4. Open nRF Logger to see/export your log.

You may also create a macro that will disable cccd after some time, or use automated tests engine to test your device. With nRF Logger closed there won't be ui updates.

jasonbu, 2016-12-06:
my main demand is **when notify is processing, disable char is avaiable**, but now, the button is not able to be pushed.we also got other debug solutions.Just make sure if it's possible to got this feature here

philips77, 2017-01-20:
Hi, did you try the version from yesterday (4.9.0)? I optimized logger a bit and now notifications are logged in bulk insert every second so perhaps it may work a bit faster. The next version coming soon or in couple of weeks may improve it a bit more.

jasonbu, 2017-02-06:
tks and it works perfectly.

closed 2017-02-06 by philips77