nordicsemi/Android-nRF-Connect #29: Fails to connect to some dual mode devices

pdjstone, 2017-02-18:
When a device advertises as supporting both BR/EDR and BLE, Android will by default prefer to connect over BR/EDR. Some devices don't support GATT in this mode, and the connection will fail. In particular, devices that use the Microchip RN4677 have this problem - nRF Connect always fails to connect to these devices. Also, connections to iOS devices seem more likely to fail when connecting in this mode. 

On Android 6.0 and later, the connectGatt method of BluetoothDevice [supports a 'transport' parameter](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int)) which allows forcing a BLE connection. On older versions of Android, this version of the method is [accessible via reflection](http://stackoverflow.com/questions/27633680/bluetoothdevice-connectgatt-with-transport-parameter). Using the TRANSPORT_LE option solves connection problems for many dual-mode devices (for me at least - it's not documented what the default TRANSPORT_AUTO option actually does - some Bluetooth stacks may prefer BLE when both modes are available)

nRF Connect should have an option to prefer BLE connections over BR/EDR (or even default to it).

philips77, 2017-02-18:
True. I'll fix it in the next version.

pdjstone, 2017-04-21:
Seems to be fixed now, thanks!

closed 2017-04-21 by pdjstone

brianrho, 2024-05-11:
Quick question please -- how was this fixed? Does the app simply adopt LE **always**, with no option to use (GATT over) BR/EDR as transport? 

Or is there some hidden option to allow users to choose transport in the app?

philips77, 2024-05-14:
It is using `TRANSPORT_LE` always.

brianrho, 2024-05-15:
That's a shame. There are practically no other scanner apps that support GATT over BR/EDR, it would've been nice if nRFConnect does.