nordicsemi/Android-nRF-Connect #30: choose device in dfu mode, sometimes it not connected with tab created and opend

jasonbu, 2017-02-20:
The UI is looks like when dfu is processing.
The connect/disconnect button is hidden and dfu button is also hidden.
It can be fix by re-enter the scan tab and choose open Tab, and the connect/disconnect button and dfu button will appear and ready to use.

philips77, 2017-02-20:
Could you post a screenshot?

jasonbu, 2017-02-21:
[image]
[image]

just like the two screenshot, it is not able to be operation without rescan and open tab

philips77, 2017-02-21:
The DFUTARG tab opens because during the DFU process your device starts to adv with address incremented by 1 so, from nRF Connect point of view, it's a new device. Also, because you have configured the GATT server in the app, it is able to detect a "new" connected device and shows it on a new tab.

There are 2 questions:
1. Why DFU TARG has 2 first bytes of the address different? Was my statement above correct and this is the same device in different mode, or it's a coincidence?
2. Why this tab is empty while it should show available services.

Could you also post the log you get on this and the previous tab?

jasonbu, 2017-02-21:
1. Why DFU TARG has 2 first bytes of the address different? Was my statement above correct and this is the same device in different mode, or it's a coincidence?
 because I chanedged the mac in application.
2. Why this tab is empty while it should show available services.
 It is why i put issue here, it seems not to happen every time. when dfu two files on single file continuesly, it happen sometimes. next time, i will catch the logfile

philips77, 2017-04-07:
Hi, I managed to reproduce the issue but still didn't have time to check the cause and fix it. I'll keep it open here as a reminder.