nordicsemi/Android-nRF-Connect #31: Ability to specify expected ATT error code

simonratner, 2017-02-24:
For read/write ops, in addition to `expected="FAIL"`, it would be great to be able to specify the expected ATT result code, and pass the test only for that result (test should still fails on any other error):
```
<write ... expected="FAIL" expectedResult="145" />
```

philips77, 2017-02-24:
Interesting. I'll look into it.

philips77, 2017-02-24:
Hi, I've just released version 4.10.0 on Google Play. It doesn't have the feature you requested for, as I didn't have time ti test it. However, here's you special build:
[nRFConnect-4.10.0-expected-error-codes.apk.zip](https://github.com/NordicSemiconductor/Android-nRF-Connect/files/799942/nRFConnect-4.10.0-expected-error-codes.apk.zip)

I hope it works. I changed expectedResult to ```expectedErrorCode```. This is only checked with ```expected="FAIL"``` and accepts integer values > 0. Timeout has code 8.
I did not test it AT ALL, but I trust my code ;) If you find some time to check it, feedback would be appreciated.

Also, mark that there are now 2 other operations available in Automated Tests and Macros: sleep-if and sleep-until. They take same attributes as for example ```write```. One sleeps until characteristic value is equal to given value (works when notifications are enabled and value can be changed, or, in case of macros, on server on writable characteristics) or timeout occur. Second one sleep until char value is not equal to given value or timeout occur.

Sorry for the delay, couldn't implement it faster... ;)

simonratner, 2017-02-24:
Seems to work, based on one very quick test. Thanks! ;)

philips77, 2017-02-28:
Hi, did you have some time to test it more precisely?

simonratner, 2017-02-28:
Yes, all good.

closed 2017-04-27 by philips77