nordicsemi/Android-nRF-Connect #32: Send FLOAT value

jgrocha, 2017-03-08:
Hi,
I would like to send a float to a BLE device (RedBear). I declared a characteristic of Double type.

```
BLEDoubleCharacteristic tempCharacteristicCalOrigin = BLEDoubleCharacteristic("CCC2", BLERead | BLEWrite);
```

On the Android nRF app, I change the value type to FLOAT 32.
When I try to write the exponent and the mantissa, the app complains with *This value does not match its type*.

[image]

I'm calculating the mantissa and exponent using the online converter [FloatConverter/IEEE754](https://www.h-schmidt.net/FloatConverter/IEEE754.html).

Using that calculator, the values for the exponent and mantissa would be 131 and 3866624 for the double number 23.375. But it does not work.

Which are the values for the Exp. and Frac. fields? How do I compute them from a double value?

Thanks!

philips77, 2017-03-08:
Hi, in BLE it is IEEE-11073 32-bit FLOAT standard used for sending Float values, and nRF Connect is using this format. Currently there is no option to select IEEE754 type but I completely understand it should.
What you could do for now is convert your value to bytes and send as BYTE ARRAY.

jgrocha, 2017-03-08:
Thanks @philips77 
I'll try to find a suitable calculator based on IEEEc11073 and I'll post it here, before closing the issue.

jgrocha, 2017-03-11:
I've ended up by using a UINT 32 value to send a float value.
To send the float value in JAVA:
```JAVA
Float value = 3.14159f;
Integer intBits = Float.floatToIntBits(value);
gattCharacteristic.setValue(intBits, BluetoothGattCharacteristic.FORMAT_UINT32, 0);
status = mBluetoothGatt.writeCharacteristic(gattCharacteristic);
```
To receive the value in C (Arduino):
```C
void tempCharacteristicCalSlopeHandler(BLECentral& central, BLECharacteristic& characteristic) {
  const unsigned char *a=characteristic.value();
  uint32_t i = 0;
  if (characteristic.valueLength() == 4) {
    // arduino is little endian
    i = (uint32_t) a[0] + ( (uint32_t) a[1] << 8) + ( (uint32_t) a[2] << 16) + ( (uint32_t) a[3] << 24);
  }
  float *f = (float*)(&i);
  Serial.println(*f, DEC); 
}
```
Thanks @philips77 for the feedback.

closed 2017-03-11 by jgrocha