nordicsemi/Android-nRF-Connect #33: Add write-no-response operation

simonratner, 2017-04-04:
Pretty self-explanatory, for writing WRITE_TYPE_NO_RESPONSE characteristics.

philips77, 2017-04-04:
On the Write dialog select Advanced (v) and there you can switch write types. Write No Respond is Write Command, available only with required property.
For Automated Tests and Macros there is type="..." attr.

simonratner, 2017-04-05:
Apologies, I wasn't clear. I am specifically talking about automated test
xml scripts.

Do you have an example of this type attr to write to a write-no-response
characteristic? I can't find it in the docs.

philips77, 2017-04-05:
Oh, sorry. Indeed i forgot to mention. But it's documented in Macros: https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/master/documentation/Macros#write-characteristic
The attribute is the same in AT. I'll update the docs soon.

closed 2017-04-07 by philips77