nordicsemi/Android-nRF-Connect #34: Automated tests: assert-value fails incorrectly

simonratner, 2017-04-10:
I have an `<assert-value value="90330000"/>` check inside an `<assert-notification>`. In the logs, I see the following:
```
Notification received from ..., value: (0x) 90-33-00-00
"(0x) 90-33-00-00" received
Notification received
Assert-Value...
Value of characteristic '90330000' is not equal to ''
```

Any idea why the assertion is failing even though it seems to receive the correct data?

simonratner, 2017-04-10:
Same seems to be true for `<assert-value>` inside `<read>` and `<read-descriptor>`.

simonratner, 2017-04-10:
App version 4.11.0.

philips77, 2017-04-10:
To bad... I'll look into it when get back from vacations.

simonratner, 2017-04-10:
👍

shafiul2s, 2017-05-06:
Yeh we need  back to

philips77, 2017-05-09:
I found the but. Sooo stupid. Will publish an update soon.

philips77, 2017-05-12:
Fixed in version 4.11.1. Hopefully. Could you verify, please?

closed 2017-05-15 by philips77