nordicsemi/Android-nRF-Connect #35: 133 error while connecting BLE Devices

harminder-emobx-com, 2017-05-11:
Hi ,

I am getting some issues with the latest Play Store Build on Samsung device.  I am getting error when i am connecting to a BLE device "Puck JS". It gave me error 133 while connecting, After a long time i was able to connect but after few seconds it disconnected. 
[Log 2017-05-10 14_31_48.txt](https://github.com/NordicSemiconductor/Android-nRF-Connect/files/991988/Log.2017-05-10.14_31_48.txt)
[image]

I am using Samsung Galaxy Note Edge with 6.0.1

philips77, 2017-05-11:
Hi, it's not an nRF Connect bug, rather the firmware you have on your Puck JS device, or an Android. Error 133 is raised when Android receives something unexpected or does not receive something expected in some time frame. Looks like your device does not follow BLE spec in 100%.
Some Android devices are more strict when it comes to timings so it may work on some other phones. The rule should be to to accept weird packets and send very strict, but the older phone is this window is moved more to the right. You may also try to restart Bluetooth or even the phone, sometimes it helps as it may be something with the Bluetooth stack on the phone. 
But most probably, there exist a finite number of characters that you could type in your Puck fw to make it work without disconnecting. It's just a matter of finding them ;)

I'm closing the issue here. If you need help to solve this problem please contact the Puck JS team, or write a question on our DevZone. Our support team is much more qualified to answer this kind of questions. Thanks!

closed 2017-05-11 by philips77

harminder-emobx-com, 2017-05-11:
@philips77  i have also try other BLE device i thought it can be issue of Puck JS but on my fitness band same issue is their. it is responding with same data. 

Can it be samsung's Bug ?