nordicsemi/Android-nRF-Connect #37: Disconnect immediately After the connection

hackill, 2017-07-17:
public void onConnectionStateChange(final BluetoothGatt gatt, int status, int newState) {}

    sometimes ,Normal connection   onConnectionStateChange(final BluetoothGatt gatt, int status, int newState)    status = 0 newstate = 2
  but  immediately onConnectionStateChange callback status = 0 newstate = 0

hackill, 2017-07-17:
[image]
see the log

This phenomenon is easy to appear on some phone.

hackill, 2017-07-17:
@philips77    i'm one of your partners from China.  This problem caused great trouble to our products. We hope you can lay emphasis on this problem ,thanks!

We have a statistics —— This problem happened 1300 times in 300 user's mobile phone.

Drolla, 2017-07-17:
With nRF Connect revision v.4.12.0 I encounter the same problem: The connection cannot be established correctly. With revision v.4.11.1 I haven't observed this issue.

philips77, 2017-07-18:
This behavior is not driven by the app. There was no significant change in 4.12 that could led to such disconnection. Just try to restart Bluetooth or the phone. Also, it may be related to how your device reacts to new SDK features (in case you changed SD or SDK), such as DLE, MTU, etc. In other words, it is either the phone, or the device. THe app just displays events it receive from the system.
You may also try downgrading the app by downloading APK from GutHub->Releases and check if it really behaves better.

hackill, 2017-07-18:
@philips77  hi philips77, Thank you for your attention.
  We are wearing equipment businesses，we also have our android app.   
  Not only the nRf connect app（4.11，4.10 ,even earlier version）,but also our app have this phenomenon.
  ``` The phenomenon is just onConnectionStateChange callback 0 2,it means connected , and than we discovery service. but suddenly the onConnectionStateChange callback 0 0 it means disconnected, so the connect become break
  ```
  see the picture above.  

  This phenomenon appears in many of China's mobile phone. We also tested our competitors, they also use the bluetooth Nordic.They also appear the same problem. This phenomenon happens sometimes, and not every time. some phone is noraml, some phone is bad.

  I hope you can understand what I say, because I am using Google translate.

hackill, 2017-07-18:
I guess it's not software, it's hardware. On the problem phone, other brands of bluetooth, without this phenomenon

philips77, 2017-11-10:
Hi, did you manage to solve this issue?
It's an old issue, but I may suggest checking connection interval. Service discovery of find using interval 7.5 ms, after which Android switched to some higher value. In your case the phone or your device disconnects as something is different then expected.

philips77, 2017-11-10:
Also, if you are using Moto:

Moto G5 phone is sending two link layer requests back-to-back without awaiting a response from the nRF, which is a violation of the Bluetooth specification, as per Core specification v4.2, vol 6, Part B, Chapter 5:
For all other Link Layer Control Procedures, only one Link Layer Control Procedure shall be initiated in the Link Layer at a time per connection per device. A new Link Layer Control Procedure can be initiated only after a previous Link Layer Control Procedure has completed.
 
Unfortunately, our S110 and S130 stack will not handle this spec-violation, but this behavior has been worked around in the S132 v4 and v5 stacks for the nRF52832 device. An option can be to switch over to the nRF52832 device.

closed 2017-12-20 by philips77