nordicsemi/Android-nRF-Connect #38: Non optimal display of the environmental sensing data with version v.4.12.0

Drolla, 2017-07-17:
nRF-Connect revision v.4.12.0 displays the the temperature and humidity data of the environmental sensing service as 2-byte hexadecimal values, which is not practical.
The previous revision, v.4.11.1, displays these 2 characteristics well as 2-byte signed value (with a resolution of 0.01 degrees Celsius). This way of displaying the temperature and humidity is much more convenient than a raw 2-byte hex value.

philips77, 2017-07-18:
Hi, are you sure you have "Parse known characteristics" checkbox checked in the menu (on the connected device screen)? I don't think I was doing anything with this service in the latest release...

Drolla, 2017-07-18:
I am pretty sure that I have checked the 'parse known characteristics', but I will double check again tomorrow.

Drolla, 2017-07-24:
I am not able to reproduce the problem with the newest revision (4.12.0). I installed 1st the previous revision (4.11.1), and the humidity and temperature values have been displayed correctly. Then I updated again to the latest version (4.12.0), and the humidity and temperature values have also been displayed correctly.
I consider therefore the issue as closed.

Drolla, 2017-07-24:
Issue is not reproducible.

closed 2017-07-24 by Drolla