nordicsemi/Android-nRF-Connect #39: Make selection of write value type sticky

nkolban, 2017-07-18:
I am using this fantastic app and I have a suggestion for the workflow.  When I am connected as a client to a BLE Server and wish to write a new characteristic value, when the dialog in which I can enter the new value appears, the type is always set to "BYTE".  In my tests, I then set this to "TEXT" and enter a string and click SEND.   Now, when I want to send a new value and open the dialog again, once more the value type is back at "BYTE" and once more I have to change the value type to "TEXT".

My suggestion is that if a user (me) changes the value type for a characteristic, the tool remember the last value type I used and makes that the current setting the next time I open the dialog in the same run of the tool.  This will save me a few steps in my workflow.

philips77, 2017-07-18:
Hi, thanks for sharing the idea. I'll check the code, perhaps it's a minor update. Btw, you may use macros if you want to repeat sending the same data.
https://devzone.nordicsemi.com/blogs/1142/nrf-connect-macros-currently-android-only/

philips77, 2017-12-08:
Hi @nkolban, I investigated your request and found a nice solution that you can use now. When you want to send (n+1)th packet, you may go to LOAD tab, click the *Auto Save* and then the EDIT icon on the right. All the fields and field types will be copied to the NEW tab. Right now you'll have to live with this...

closed 2017-12-20 by philips77