nordicsemi/Android-nRF-Connect #40: Android O Support

JBlaz, 2017-07-27:
There are some devices that are not picked up by the scan on both my Pixel and Nexus 6P with Android O. The same devices scan just fine with Android Nougat and other Android versions.

philips77, 2017-07-27:
Could you share the adv packet of those devices obtained on Nougat (raw data or log from nRF Logger)? I'm testing on Pixel with O and don't have such problems.

JBlaz, 2017-07-28:
[image]
This device IS visible on a Nougat phone and is not shown on my Pixel running O.

philips77, 2017-07-28:
Stupid question: you don't have any filter set on Pixel?
Are the scanner setting set to default? (Continuous scanning ON, mode: low latency, offloaded scan batching OFF)

JBlaz, 2017-07-28:
Definitely not a stupid question but I have no filters set on the pixel.

cmmouritsen, 2017-11-10:
Any update on this? I am having a similar issue. I have a Moto G (2nd Generation) that is running Marshmallow that picks up my beacon fine in nRF Connect, but on my Nexus 6P running Oreo, nRF doesn't find it. Any suggestions?

philips77, 2017-11-10:
If a device advertising with invalid data it is ignored by Android O. Invalid data, unfortunately, means also 0s after the last record, which may be a bug on Android. If you have access to a sniffer of to device fw check the adv data and make sure there are only valid EIRs (valid EIR has 1byte for length, 1byte of type and len-1 bytes of data). In theory zeros at the end are ok, but Android ignores such packet.

cmmouritsen, 2017-11-10:
Thank you, @philips77. My data looks valid to me. Do you see anything that looks off?:

[image]

philips77, 2017-11-10:
You don't see those additional data on older Androids. You'd have to use sniffer or debug the fw. Perhaps checking hci log from Android could help if you don't have sniffer. On Developer settings turn on hci snoop log, scan, turn it off and export log to PC. Since Nougat it's inside bug report and can be extracted somehow. Open this log in Wireshark.

JBlaz, 2017-11-10:
I have learned that on my Pixel 2 running 8.0 I can again scan for these invalid entries. Hopefully this means that in an update the pixel 1 will be able to see them again also. Thank you @philips77 for finding the culprit.

closed 2017-11-10 by philips77