nordicsemi/Android-nRF-Connect #44: Write with response results in disconnection on Nexus 7 Android 5.1

lilliesAndRoses, 2017-09-06:
Issue:
On Nexus 7 tablet running Android 5.1, if I send a "Write request" (write with response), I get GATT error (0x85) followed by GATT CONN TIMEOUT (error 8), resulting in disconnection of the device. The device to which this is being sent is using NRF52832 and SDK 13.
If I send a "Write without response", data does get sent, but I am left at the mercy of Android. It would send it at its leisure, and this is not good for my app.

It works fine on Samsung tablets with builds 4.4, 6.0.1, and 7.0.1.

Not sure if the issue is related to Nexus 7 or Android 5.1, but the problem is blocking usage of this platform for my app.

Thanks.

lilliesAndRoses, 2017-09-12:
Does anyone bother about Android 5.* anymore?

philips77, 2017-09-12:
Hi, this if course should work on Nexus 7. Did you try SDK 13 or some older one? I didn't experience such a problem on this tablet, but as you noticed, Nexus 7 is very old and it (as well as Nexus 4) had a ble chip with bad performance. We don't usually develop on them. The problems we had back then were related more to losing pairing, or just happened at random times (at least we didn't see any pattern). Your issue seems to be repeatable and it's worth investigating. Do you know if the samples from SDK 13 which have WWR also fail on this tablet?

lilliesAndRoses, 2017-09-13:
Thanks for your response.
Could not reproduce the problem with SDK 13 sample. Would look in our firmware code.

closed 2017-09-13 by lilliesAndRoses