nordicsemi/Android-nRF-Connect #45: Wish list: Ability to provide descriptions for custom UUIDs

nkolban, 2017-09-24:
When I work with devices that have custom services and characteristics, these have generated UUIDs.  I would love to be able to "describe" these to nRF Connect such that when I view them in the tool, they have more descriptive names.  For example, if we look in the following screen shot, we see "Unknown Service".  I would like to able to provide my own descriptions.  In an ideal world, we would have a database hosted on the Internet that contains a list of services that could be consulted with the ability for users to add/update their own.

[image]

philips77, 2017-09-25:
Hi,
You may already do this. Long click a service or a characteristic and select the EDIT icon on the action menu. For characteristics you may also check format=text field so the values are parsed in ASCII.
As to the global database in the Internet, we have it on our backlog. Currently the app does not use Internet permission so it could be safely used by customers with secret projects so we need to make sure the positive pluses overwhelm negative pluses... :)
[image]

philips77, 2017-09-25:
In the navigation drawer you\ll also find a menu item "Definitions" where you'll find all your custom name + those defined by SIG or other known to the nRF Connect. You may share the definitions between youy phones. The definitions file is also compatible with nRF Connect for PC.

closed 2017-09-25 by philips77