nordicsemi/Android-nRF-Connect #46: Connect Often Falied

kcjian, 2017-10-14:
I wrote a new android application,when I'm going to connect  some device, often get a error code such as 133 or 129(HUAWEI mobile phone is more obvious),but the nrf connect success probability  is very large. So,I want to know what special treatment do you have? And can you give me some good advice? Thanks very much.

philips77, 2017-10-14:
Hello,
We really aren't doing anything special except:
1. There are some delays added here and there to give the Bluetooth adapter more time despite receiving the callback. For example after getting onConnectionStateChanged before calling discoverServices(), as far as I remember. I log then on Debug level in the log so you know where and when.
2. On older versions of Android I was calling a hidden method gatt.refresh() to clear device cache after disconnecting, before calling close() on the gatt object. So that the next time you connect the phone does a full service discovery instead of using cache. This for some strange reason helped (perhaps because of some extra time required for sd).
3. We highly recommend using the BleManager class from nRF Toolbox. You'll find it in profile package and the app on GitHub. The code there is quite similar to what's in nRF Connect.

closed 2017-11-10 by philips77

elsenpan, 2018-01-18:
@kcjian How did you resolve the issue finally?

kcjian, 2018-01-18:
Sorry,I'm can't resolve that always some mobile phone has problem, just try reconnect.

nevergoneok, 2018-06-16:
@philips77 On older versions of Android?Can you tell which version is it?

philips77, 2018-06-16:
I call refresh () after disconnection on Android 4.3 - 5.1. since 6 it's only on demand. But that's just based on some observations. Doesn't mean that 6+ is without this error, but it happens less often.
133 in my opinion happens when something in the connection was unexpected, for example a packet wasn't received in a window, was receiver when shouldn't, contained some invalid data, etc. The phone should tolerate a lot and send strict, but the first versions weren't tolerate and had bugs themselves. In 6 Google has rewritten a lot, in 7 more, and added a lot of tests.