nordicsemi/Android-nRF-Connect #47: Automated test parameterization of write value

frubenas, 2017-11-10:
Hi,
in an automated test xml file, I am unable to use a definition in the value field of a write request to write a byte array.
For instance, I define a VAR value as it follows:
`<set name="VAR" value="00010203" />`
Then in the test definition I write:
`<write description="Write a value" type="WRITE_REQUEST" service-uuid="${MY_SERVICE_UUID}" characteristic-uuid="${MY_CHAR_UUID}" value="${VAR}" />`
When running the test I get:

> Parsing test suite failed:
> '${VAR}' is not a valid byte array

What am I doing wrong?

philips77, 2017-11-10:
Hmm... Let me check it on Monday.

philips77, 2017-11-24:
Hi, it will be fixed in the next release (4.15.1 or 4.16). So far `value` and `value-string` could not be parametrized. Reason, as far as I remember, as to validate HEX before execution, not during. But having an option to parametrize them is more useful than validation, so I decided to change it.

philips77, 2017-12-08:
Released in 4.16.0, I guess. Did you try it?

closed 2017-12-08 by philips77