nordicsemi/Android-nRF-Connect #50: Change complete local name of an advertiser

saintnever, 2017-12-07:
I'm wondering how to change the complete local name of an advertiser. In my Huawei P10 plus the name is "Server 7 places available VKY-AL00", which just leads to "Error1: Advertisement Data too large"

philips77, 2017-12-07:
Actually you'll be able to do this in the next version that's coming today or tomorrow.
Right not you have to go to Android Settings->Bluetooth->Device name.

saintnever, 2017-12-08:
Nice! Looking forward to the new verison.

philips77, 2017-12-08:
Released in 4.16.0. On Advertiser tab there is now an icon with a pen in the menu.
Keep in mind, that it changes the name of the phone/tablet, so it will change the Complete Local Name value for all advertising packets.

closed 2017-12-08 by philips77