nordicsemi/Android-nRF-Connect #51: Detect all devices connected

vibexie, 2017-12-07:
Hello, nrf-connect-android will auto connect the device which has been connected by my device, so how to make it?

philips77, 2017-12-07:
Hi,
The current version or nRF Connect will automatically connect to connected devices when you enable the GATT Server. Go to navigation menu -> Configure GATT Server -> Click a drop down menu at the top and select anything except Disable. It may be an empty (create a new configuration) configuration, but has to be enabled. Then go back to the main screen and it should open tabs for each connected device immediately. You also have to have the *Settings -> Connectivity -> Show incoming connections* option enabled (default).

Kepp in mind, that nRF Connect (and other apps that connect to devices using GATT server) will hold a strond reference to the physical connection, so if you disconnect the device from your app it will still be connected in nRF Connect. This has changed in Android 8.1 with [this](https://android.googlesource.com/platform/frameworks/base/+/a355e5efaf45a534ee6437aa4bae7d30f18c0ec2/core/java/android/bluetooth/BluetoothDevice.java#1861) method. Since 8.1, when a connectino is found, nRF Connect will connect using `opportunity=true` so when you disconnect in your app, nRF Connect will also get disconnect event.

In some next release I'll make it less annoying (right now nRF Connect while in bg may take over the connection and hold them from being disconnected). The tab will open, but nRF Connect will not connect until requested.

philips77, 2017-12-07:
About how to make it:
Open GATT Server using https://developer.android.com/reference/android/bluetooth/BluetoothManager.html#openGattServer(android.content.Context,%20android.bluetooth.BluetoothGattServerCallback). You'll get `onConectedStateChanged` for each device already connected and that will connect while the server is open.

closed 2017-12-07 by philips77