nordicsemi/Android-nRF-Connect #52: GATT Characteristics not recognized

beegee-tokyo, 2017-12-20:
I am using GATT characteristics from [Bluetooth Specifications](https://www.bluetooth.com/specifications/gatt/characteristics), specifically 0x2A17 (Temperature Celsius), 0x2A6F (Humidity), 0x2A3D (String) and 0x2A46 (New Alert) on my BLE device.
All GATT characteristics are shown in nRF Connect, but shown as "Unknown Characteristic" and "Unknown Descriptor".
I hoped there is support for standard GATT characteristics in the application. Or maybe I am doing something wrong in the definition of the characteristics and descriptors.
Any advice?

philips77, 2017-12-20:
So... 0x2A17 is Time Update State (Guess you meant 0x2A1F), Humidity should work, 2A3D I'm adding right now (among others), and New Alert also should work. Could you paste here the screenshot from the app?
There is support for those listed in menu->Definitions.

philips77, 2017-12-20:
I've just released version 4.17.0 which should support all mentioned characteristics. Could you verify when you get the update?

beegee-tokyo, 2017-12-21:
Updated to the latest version, still the same. I have to admit that I am not testing a nRF chip, but the BLE of an ESP32 module.
I guess the problem is within my device and my descriptors. Have to learn how to correctly setup the characteristics and descriptors.
Here a screenshot of the app together with the debug messages.

[image]

philips77, 2017-12-21:
You have to set those uuids to characteristics, not descriptors. Those characteristics don't need descriptors unless they are notifying or indicating, then the CCC descriptor is required in such char. Right now your characteristic have 128bit random uuids.

beegee-tokyo, 2017-12-21:
Thank you very much. I am still learning about BLE stuff. Now my ESP32 sends correct data within correct characteristics.

Your tool is really very good to learn and debug BLE apps!

Merry Christmas and a Happy New Year!

[image]

closed 2017-12-21 by beegee-tokyo