nordicsemi/Android-nRF-Connect #53: Not Opening in Android Studio

VinayTShetty, 2018-01-06:
**Hi i extracted the zip file of Android-nRF-Connect,But its not opening in android studio.
Have attached the Screen shot 
ScreenShot1 -->ExtractedFolder.
ScreenShot2 --->Inside ExtractedFolder.
AndroidStudio_ScreetShot --->Its not Showing the s
[image]
[image]
[image]

ymbol of android studio to open.**

philips77, 2018-01-06:
That's true. As stated on the Readme, the source code of this as it's not available. On GitHub there is just some documentation.

closed 2018-01-06 by philips77

VinayTShetty, 2018-01-06:
@philips77 
@roshanrajaratnam 
Can u please Provide any source code link which will send/Receive command to BLE.
Am running in loops,As of now i have found only Scan,Connect,and Read Charcterstics.
I want the source code which will Send some commands to BLE.
It would be a Great Help,Thanks in advance.

philips77, 2018-01-06:
Check out https://github.com/NordicSemiconductor/Android-nRF-Blinky. This is a good project to start with BLE on Android. This project also depends on the Android-Ble-Library, like Toolbox.

philips77, 2018-01-06:
You'll find there BlinkyManager (https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java) class which does what you need.

VinayTShetty, 2018-01-06:
@philips77 
Thank you for Providing the link.
But i have few Error after opening in android Studio Can u provide some guidance on Resolving it.
ScreenShot1 -->After Extracting zip file
ScreenShot2 -->Inside Android-nRF-Blinky-master
AndroidStudio_ScreenShot -->Showing the error.
[image]
[image]
[image]

philips77, 2018-01-06:
Try opening first the Ble-library project only in the AS. It will create all required iml files. Then ensure your settings.gradle file in the Blinky project the same as here: https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/master/settings.gradle.
In this settings file there should be a path to the ble.iml file that was created in the first step.

VinayTShetty, 2018-01-06:
I deleted the entire Project and extracted the zip file.
Am unable to find the Ble-library project folder.
[image]
[image]

VinayTShetty, 2018-01-06:
@philips77 
I checked in Setting also.But the issue is not yet resolved.
[image]

philips77, 2018-01-07:
Just like in nRF Toolbox, you have to clone the Android-BLE-Library project from GitHub, ensure the folder name is equal Android-BLE-Library, import it to Android Studio, then clone the nRF Blinky project to the same folder as you cloned the library and import as well it to AS. You had it correctly before, just had to import the library project, as I wrote.