nordicsemi/Android-nRF-Connect #54: Can't enable notifications

mattkang, 2018-02-20:
I am building a device, and I've been running into some trouble with nRF Connect. Specifically, I cannot receive notifications on Nougat or Oreo. Marshmallow and earlier work just fine. I can confirm I am able to write to the notification descriptor on the device through the app, but onCharacteristicChanged() is never called.

I am trying to pinpoint whether the issue is with Android or my device, and it's been driving me nuts.

Any and all help is greatly appreciated!

philips77, 2018-02-20:
Hi, it work for us, so perhaps your device? Try with a sample app, like HRS from the SDK. Are you sure you're sending a valid packet? Proper size, etc?

mattkang, 2018-04-09:
Hi just an update: I was able to track down the bug, and it was indeed something (stupid) on our end. We were trying to send a notification to the ClientConfig instead of the actual characteristic. Thanks!

closed 2018-04-09 by mattkang