nordicsemi/Android-nRF-Connect #56: How to detect whether location is required to be enabled on Android M?

piotr-pawlowski, 2018-03-12:
On some devices (ie Lenovo) nRF Connect prompts to enable location. On Samsung Galaxy A5 or S7 location is not required (location permission is required of course). How do you detect BLE requirements?

philips77, 2018-03-12:
Hi,
There is no any API to check it. In my opinion it should be always required, but as you know some phones (e.g. all Samsung phones) don't require location to be enabled.
So what we do is we show a message that location is required and start scanning at the same time and if a packet is found we hide the message and save in shared property that location is not required on this phone.
Check nRF Blinky on GitHub for source code.

piotr-pawlowski, 2018-03-12:
I supposed that it works in that way. Thank you for the response!

closed 2018-03-12 by philips77