nordicsemi/Android-nRF-Connect #57: Connection with Samsung Galaxy S5 Neo not possible

schoensn, 2018-03-26:
Hi,
I want to try to connect a Samsung Galaxy S5 Neo (Android 6.0.1) with nRFConnect App with the HeartRateService example. The HRS runs on a nRF52840 Preview DK with SDK14.2 and s140_nrf52840_5.0.0-2.alpha_softdevice. The connection procedure always fails.
With my tablet Nexus9 with Android 7.1.1 all works fine... 
Attached I have a wireshark log of the connection process. Have you any idea, why the connection fails with my mobile phone? 
Best regards, 
Stefan
[182603-samsung_s5_neo_connection_fail-hrs-mot-sbst.zip](https://github.com/NordicSemiconductor/Android-nRF-Connect/files/1847599/182603-samsung_s5_neo_connection_fail-hrs-mot-sbst.zip)

philips77, 2018-03-26:
Hi, could you post it on DevZone? It's related to our SDK rather then nRF Connect. The problem might be with new Bluetooth features that your phone fails to ignore if doesn't support, but we'd have to get sniffer logs to know exactly.

schoensn, 2018-03-27:
Hi,
thanks for your reply..I have already attached the sniffer logs..  Yes I
can post it  in the devzone too.
Best regards Stefan

2018, 22:30:

closed 2018-09-17 by philips77