nordicsemi/Android-nRF-Connect #58: Characheristic Value Representation

Skodra, 2018-04-03:
I think it would only be adequate to take in account _Unit_ value of the _Characteristic Presentation Format_ descriptor and base on that format the _Value_ field of the characteristic. Example,  I have set all _Characteristic Presentation Format_ fields accordingly  but _nRF Connect_ would not display value of characteristic as unsigned 16-bit integer.
[image]
Issue posted here also https://devzone.nordicsemi.com/f/nordic-q-a/31253/is-it-possible-to-display-integer-on-nrf-connect-app

philips77, 2018-04-10:
Fixed in 4.19 :)

closed 2018-04-10 by philips77

Skodra, 2018-04-26:
Thanks for the quick response! Was this already in you plans?

philips77, 2018-04-26:
No, but it was a fairly small but useful feature, so we decided to go with it quickly. Hope it works well.