nordicsemi/Android-nRF-Connect #59: Blood Glucose Logging Incorrect

nateridderman, 2018-04-10:
When I read a blood glucose measurement characteristic from a BGM (via RACP), nRF will output a friendly log message with the parsed values. However, it parses the sample location and sample type incorrectly. The type is the lower nibble and the location is the higher one, but they are swapped in the log output. The nRF Toolbox app parses it correctly: https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/gls/GlucoseManager.java#L226

Relevant BLE spec (without clear explanation of which nibble is lower) : https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.glucose_measurement.xml

philips77, 2018-04-10:
That is very likely. I did both of the apps hoping at least one will be ok. My test data had 0x11 there to make sure both will pass test ;)

Expect a fix in the next version.

philips77, 2018-04-11:
Fixed internally.