nordicsemi/Android-nRF-Connect #60: Automated tests: String assertion fails incorrectly

NikBrg, 2018-04-13:
I have an `<assert-value value-string="GB"/>` check inside a `<read>`. In the logs, I see the following:

```
Read response received from ..., value: (0x) 47-42-00
Asserting characteristic value...
Value of the characteristic 'GB' is not equal to 'GB'
```

Any idea why the assertion is failing even though it seems to receive the correct data?

philips77, 2018-04-13:
There is a 00 at the end. "GB" has just 2 bytes, but 3 we received.

NikBrg, 2018-04-13:
Ok thank you! How do I add the string terminator 0x00 in the XML file? At the moment the assert looks as follows:

`<assert-value description="Check country" value-string="GB" />`

philips77, 2018-04-13:
Well... You may do
```<assert-value description="Check country" value="474200" />```

NikBrg, 2018-04-13:
This is what I wanted to avoid... Does that mean the string assertion `value-string`  cannot be used as documented?

philips77, 2018-04-13:
If the value returned wouldn't be null-terminated then it would work. I can 'fix' it (add support for null-terminated during values) in the next release.

NikBrg, 2018-04-13:
Thank you, that would be great! Actually the only reason we are using string assertions is, because we are only interested in the data before the null termination (and not afterwards).

philips77, 2018-04-13:
So it may be that you have more non-zero bytes there and just want to check 'until 00'?

philips77, 2018-04-13:
Just write some examples that you'd like to have in the test. I'll come back to it before next release.

closed 2018-05-15 by philips77