nordicsemi/Android-nRF-Connect #62: Possible bluetooth stack corruption

andres7293, 2018-04-26:
Hi.
Using the nrf connect app with Huawei mya-L11. Under certain actions, the phone needs to be restarted in order to work again.
Here are the steps:
1. Open nrf connect.
2. Scan devices.
3. Connect to desired device. 
4. Turn off bluetooth and wait few seconds.
5. Turn on bluetooth.
5. Try to connect again with device.
6. Can't connect to device anymore.

At this moment, the phone needs to be restarted to work again.
The screenshot below shows the error.

Thanks !!!.
[image]

andres7293, 2018-04-27:
Hi again.
Doing the same process previously explained. When device can't connect, going to Bluetooth settings and doing bonding process with device, independently of the result, opening nrf connect again, the connection has been established without user interaction.

This seems Android bug, rather than Nordic application.

philips77, 2018-04-27:
Yes, looks like an Android bug or a bug in your phone's controller. Try a different phone.

closed 2018-04-27 by philips77