nordicsemi/Android-nRF-Connect #67: Implementation of refresh services / clear cache.

Domvel, 2018-06-07:
I see that this is not open source. But I hope for a support about the implementation of "Refresh services". 
How did you implement this in native code? I can't find any method of the android bluetooth API.

For my app I have the same issue. When the peripheral changed his services / characteristics some android devices has a cache problem after connection. And my current solution is to reset the bluetooth enable state. (off / on) 

If you don't want to publish any line of the code, I'm also happy with a tipp. 😃

philips77, 2018-06-07:
Check out the implementation here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1401
In DFU we have the additional parameters `force`, but you get the point.

Domvel, 2018-06-07:
@philips77  Thank you very much. I'll test this out.

closed 2018-06-07 by Domvel

philips77, 2018-06-07:
Keep in mind that it's asynchronous method and it may take some time before it's complete (and of course you don't have any callback for this). After the cache is clear, and you're still connection, the Android will initiate internal service discovery which will finish after another some time, depending on the connection interval and number of attributes to discover. You will not get onServicesDiscovered when it's done. When you'll call `gatt.discoverServices()` before it's done, it will abort and start discovery again. When after it's done, it will result immediately with cached services.

Domvel, 2018-06-07:
Thanks for the infomations. My first test looks good. For now I have to call the method manually. Like in nRF Connect. Later I want to invoke this automatically after a connect. But as you wrote, this method has no complete callback. Maybe I have to play with a timer. :/ 

1. Connect to device.
2. Refresh device cache. (wait a few seconds)
3. Re-Connect (Disconnect -> Connect)

Ok usually refreshing device cache is not required on each connect. Only when l.e.g. peripheral services / characteristics has changed. I may only call this on a software update via the app. BUT this does not fit the case if the peripheral was updated from another device. 

In my opinion Android should have an optional argument of the connect method to force a refresh device cache. What you think?

philips77, 2018-06-07:
The spec is clear about cache. For not bonded devices master should not cache services if a device has Service Changed characteristic. Android ignores this rule and always caches.
For bonded devices it should always cache and clear the cache in SC indication, which Android does correctly, but if the app calls `gatt.discoverServices()` in `onConnectionStateChanged(..)`, which is called before resuming security and before this indication is sent, it will get old services. For bonded devices I would recommend a delay before discovery to ensure SC ind was received, like we do in the BLE Library.

Domvel, 2018-06-07:
Im my case the device is not bonded. (wouldn't work at this moment.) Ok that explains it, that Android ignores this rule and always caches. Interesting. Anyway... connect should have a clear cache option.