nordicsemi/Android-nRF-Connect #82: Auto-select text input when CPF descriptor with UTF-8 is present

carlescufi, 2018-09-18:
When a Characteristic contains a Characteristic Presentation Format that contains UTF-8 as its value, nRF Connect Should switch to text input instead of hex, since the contents of the characteristic value are supposed to be UTF-8 encoded text.

CC @mostafaberg @jovirus @roshanrajaratnam @philips77

philips77, 2018-09-18:
Easy. Had you written this 1h earlier, you'd have had it in the released that just went public :)

philips77, 2018-09-18:
The problem is with other values of this descriptor. Why utf8 will have a string input while other values a generic one. But I understand your use case. Let's say this is the most common.

philips77, 2018-09-19:
Released in version 4.20.2.

closed 2018-09-19 by philips77