nordicsemi/Android-nRF-Connect #83: Known log file size

mattmaclennan82, 2018-10-24:
I'm about to start a series of long duration tests and I want to use nRF Connect to record the data from my Bluetooth device. The test duration is 60 hours with notifications every second on 3 services.

Is there a known limit to the log file size?

philips77, 2018-10-25:
Well.. it may not be possible to share the log afterwards (using Share button), but saving to a file should work. Install nRF Logger app so that even if nRF Connect crashes, the log is saved there.
For such very long tasks I would actually recommend writing a small app, perhaps based on nRF Blinky, and log fro it, as you have then control over what's being logged and how. nRF Connect is using Content Provider, which is not the fastest way of doing it, so it may become laggy (?). Bug 3 notifications each second should not make it unusable.

I'm really curious about your finding. Could you please share if the app survived? Thanks!

mattmaclennan82, 2018-10-25:
Thank you very much for the quick reply!

I'll keep you updated. nRF Connect has been invaluable during diagnosis and development work. I often refer to it as "the grown up app", whenever ours crashes or generally disappoints!

Matt MacLennan

philips77, 2018-11-06:
Hi, do you have any findings?

mattmaclennan82, 2018-11-06:
Sort of…

We have an 86 hour long test that is sat “somewhere” on the Android tablet. We can’t find it as a file, but we can see it in nRF Logger. It takes a little while to load the log file. If we try to SAVE, COPY or SHARE the log, it fails and the “Not Responding” dialogue box appears.

We have managed to recover a 17 hour long test though (https://stevevick-my.sharepoint.com/:t:/p/matthew_maclennan/Ef4sHn8A6C1DnTEjWBg2GoEBx-bcx2vM54iasLP0TLVu8g?e=RD787j)

I had a go with nRF Cloud to see if there was any remote logging facility. I love the API exposure, but there’s a gap where logging should be. I’ve been in touch with Markus about the possibility of adding that as a feature.

Let me know if there is something clever I can do to recover the 86 hour log.

Thanks,

Matt

closed 2022-08-12 by philips77

omidxs, 2022-12-30:
Hi
Im new to BLE and I have the same problem with nRF logger. do you have any findings?